bambulab/BambuStudio #6133: Send to Multiple Device wrong AMS slots utilized

CepheusSolutions, 2025-03-22:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

Bambu X1C

### How to reproduce

Go to 'Send to Multi-Device"
Select Use AMS
See that the three colors used for print are all labeled A1
Select a color and see you cannot select any different slots in the AMS
Send to Printers and confirm that all colors are printed from A1 slot and not from slots 2,3,4.

### Actual results

[image]

[image]

[image]

[image]

[image]

### Expected results

Sending to multiple devices should allow the proper use of AMS slots preselected in Bambu Studio

### Project file & Debug log uploads

[Cookie Key Chain v2.zip](https://github.com/user-attachments/files/19405794/Cookie.Key.Chain.v2.zip)

[log.zip](https://github.com/user-attachments/files/19405801/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

CepheusSolutions, 2025-03-28:
The very latest version is still broken.

[image]

[image]

jup2000, 2025-04-08:
With this version, sending to even SINGLE machines on my print farm almost always prints with the WRONG slot.

walterwongbbl, 2025-04-16:
Hi. @CepheusSolutions 
Please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

CepheusSolutions, 2025-04-26:
Issue resolved thanks.

closed 2025-04-26 by CepheusSolutions