bambulab/BambuStudio #6135: CAN'T LOGIN VIA GOOGLE TOKEN (facebook also not working?)

echoround, 2025-03-24:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Try to login, Europe region, via google token

### Actual results

"Site can't be reached" and url is this: 

http://portal:3000/api/v1/user-service/user/tplogin?source=google&callback=https%3A%2F%2Fbambulab.com%2Fapi%2Fauth%2Fcallback%3Fsource%3Dgoogle%26locale%3Den-us%26redirect_url%3Dhttp%253A%252F%252Flocalhost%253A13618%26from%3Dstudio

### Expected results

Login to bambu via google

### Project file & Debug log uploads

http://portal:3000/api/v1/user-service/user/tplogin?source=google&callback=https%3A%2F%2Fbambulab.com%2Fapi%2Fauth%2Fcallback%3Fsource%3Dgoogle%26locale%3Den-us%26redirect_url%3Dhttp%253A%252F%252Flocalhost%253A13618%26from%3Dstudio

### Checklist of files to include

- [ ] Log file
- [ ] Project file

piRsquaredH256, 2025-03-24:
Same on my side. Can't Login anymore. Tried on two different Networks - On two different Machines. Both running Mac OS Sequioa 15.3.
Sign in via Google, Facebook, Apple and System Browser yields the same result.

darth-pika-hu, 2025-03-24:
@piRsquaredH256 I was going to reply to you with the fix. But seems you are the same person on reddit. LOL

sewalshmsft, 2025-03-24:
> I get a reply that says {"error_msg":"404 Route Not Found"} if I replace the portal:3000 with api.bambulab.com

You should replace it with bambulab.com, NOT api.bambulab.com

darth-pika-hu, 2025-03-24:
@sewalshmsft You need to replace portal:3000/api with api.bambulab.com

sewalshmsft, 2025-03-24:
Either way works

Creasento, 2025-03-25:
I also have this problem. and also solved using https://api.bambulab.com/v1
but why it happen? Do I need to do this for every login?