bambulab/BambuStudio #6137: A1 mini 在盘中 Shift+A自动布局时，位置中心偏右

rushairer, 2025-03-25:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Additional system information

在win/mac下，跟版本无关，一直存在这个问题。

### Printer

Bambu Lab A1 mini

### How to reproduce

1.新建工程
2.导入stl文件
3.按下 Shift+A

### Actual results

[image]

### Expected results

应该居中

### Project file & Debug log uploads

[debug_Mon_Mar_24_13_44_11_27936.log.0.zip](https://github.com/user-attachments/files/19439633/debug_Mon_Mar_24_13_44_11_27936.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

tangketan, 2025-03-26:
这是自动摆盘的feature：A1 mini是单悬臂结构，靠近立柱的位置一般精度和稳定都会更好，所以自动摆盘会尽量将对象放在靠近立柱的位置。如果你想改回正中间，可以将这个“最佳对象位置”参数改成0.5

[image]

rushairer, 2025-03-26:
谢谢 @tangketan

closed 2025-03-26 by rushairer