bambulab/BambuStudio #6146: Mapping Filaments Issue

jdengland1, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.3.2

### Additional system information

3 GHz 6 core intel i5

### Printer

A1/A1 Mini

### How to reproduce

Slice plate
Print Plate
Try to remap colors and you can't

### Actual results

When you try to print files it will not allow you to map colors through the print service. All colors have to be mapped before printing and maybe you will get the software to map it correctly. 

### Expected results

Slice plate
print plate
remap colors as desired

### Project file & Debug log uploads

Not applicable as it is an issue in the software. I can still print however can't remap the colors. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

KolemanIreland, 2025-03-25:
Same issue here.

Apple M2 Pro
32 GB
15.3.2 (24D81)
A1 / A1 Mini / P1S / P1P

JohnSmithCYM, 2025-03-26:
Thanks for your feedback and we want to know more about this problem. Can you give a screen capture in gif or video to show the whole process?
You download ScreenToGif software and record a GIF.
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

fahaaaxiki, 2025-03-26:
Sorry, it is a bug, but we will fix it soon. Before the updating, you can solve it temporarily by doing  these steps:
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally. 
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.
[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00