bambulab/BambuStudio #6148: P1P with Upgrade Kit no longer accepted as P1S

bjoernmayer, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.2 (24D81)

### Printer

Bambu Lab P1(P/S)

### How to reproduce

1. Get P1P
2. Upgrade P1P with official kit to P1S
3. Try to choose P1S as your printer in Bambu Studio
4. Try to sync AMS colors

### Actual results

Error is shown:
[image]

Before version 2.0 this was absolutely no problem.

Why do I want to choose P1S?
- I want to use all features of the P1S. That's why I installed the UPGRADE kit. I am afraid, that all extra fans won't be used, when choosing P1P as printer

### Expected results

Let me handle an upgraded P1P like a P1S

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MedoG27, 2025-03-26:
I have your same problem

Haidiye00, 2025-03-27:
Hello，We have optimized the behavior in the internal version：p1s and p1p are compatible machines， If your preset is p1s and the connected printer is p1p, both can initiate AMS mapping and printing tasks.
We have fixed it and will release a new version as soon as possible~

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00

mbaran5, 2025-06-11:
I am seeing a regression of this error in the latest version of studio. I have a P1P with the upgrade kit. 

[image]

Haidiye00, 2025-06-26:
> 我在最新版本的 Studio 中看到了此错误的回归。我有一个带有升级套件的 P1P。
> 
> [image]

Hello, may I ask which version you are using?