bambulab/BambuStudio #6149: Upgraded P1P to P1S shows up again as a P1P

ericdano, 2025-03-25:
### Bambu Studio Version

2.00.00.95

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.4

### Printer

P1P that has the official P1S upgrade kit installed

### How to reproduce

Load up program, printer shows up as P1P

### Actual results

Shows up as P1P

### Expected results

Should show up as P1S like in previous versions of the software

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bjoernmayer, 2025-03-25:
Duplicate of #6148

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00