bambulab/BambuStudio #6150: No AMS for Filament Calibration

Radar1210, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S with AMS

### How to reproduce

1. Go to Calibration
2. Go to Manual Calibration
3. The AMS is not displayed on Flow Dynamics. Only external filament is visible.

[image]

### Actual results

Show only

[image]

### Expected results

Filament from AMS should be selectable.

### Project file & Debug log uploads

No Project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

chadteamrrc, 2025-03-25:
Same issue here

zhimin-zeng-bambulab, 2025-03-26:
@Radar1210  Could you try to let AMS feed first.

Radar1210, 2025-03-26:
@zhimin-zeng-bambulab OK, if I load a filament from the AMS, then I can select the filaments.
It would be fine if it didn't load it.

zipinDupree, 2025-03-27:
With Bambu Studio you need to go to device and load a filament first. Then the AMS will show up.

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab