bambulab/BambuStudio #6153: Multi-device no longer shows time remaining or % completed

Shpigford, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15.3.2

### Additional system information

[image]

Prior to v2, the Multi-device page would show various progress indicators for each printer (time remaining, % complete and a progress bar).

With v2, that has disappeared and it simply shows the status of each printer.

### Printer

All Bambu Lab printers

### How to reproduce

1. Go to "Multi-device" tab in Bambu Studio
2. View the list of printers
3. See that there is no progress detail

### Actual results

[image]

### Expected results

I expect to see progress details.

### Project file & Debug log uploads

It's not project-specific. Happens on all files.

### Checklist of files to include

- [x] Log file
- [x] Project file

walterwongbbl, 2025-04-21:
@Shpigford @StineDEk 
This should be a functionality regression, and we have added it to the fix list. Thank you very much.

Jmknights93, 2025-05-22:
@walterwongbbl Do we have a timeline when this will be fixed? It's been a few months now.