bambulab/BambuStudio #6156: Cannot perform FLow Calibration with filament from the AMS with Bambu Studio 2.0.0.95

michaeltavel, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD 5950x, 64g RAM, RTX 3080

### Printer

Bambu Labs P1S

### How to reproduce

Go to the 'Calibration' tab and select 'Manual Calibration'.

You are unable to select any filament from the AMS, only the External Spool is displayed as an option:

[image]

All the AMS filament is properly configured:

[image]

### Actual results

Cannot perform a Flow Calibration with filament from the AMS

### Expected results

I expect to be able to select filament from the AMS for a Flow Calibration.

### Project file & Debug log uploads

No project files or log files needed.  This can be replicated without any project loaded.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FjolnirsBath, 2025-03-25:
Receiving identical error with version 2.0.0.95 and my X1C.

michaeltavel, 2025-03-26:
I loaded the ASA filament (slot #2 in my AMS) from Bambu Studio and then went to the 'Calibration' tab:

[image]

I was able to select the filament from the Flow Calibration screen now (was no longer showing only the external spool):

[image]

But the calibration print never started and after a few minutes with no activity (and the hotend and bed temperature up to printing temperature) I got an error "Failed to get AMS Mapping Table".  I am able to print objects from 3MF project files, but not from the Flow Calibration.

[image]

[image]

michaeltavel, 2025-03-26:
After I successfully printed a project from a 3MF file:

[image]

The AMS automatically unloaded the filament from the printer.  

After this, I went back into the Flow Calibration and saw that I could only pick the external spool again:

[image]

zhimin-zeng-bambulab, 2025-03-26:
[image]

Could you share your log files when "Failed to get ams mapping tab".

In addition, we will improve the UI in the future. Thanks for reporting this issue

michaeltavel, 2025-03-26:
The error occurred at 2025-03-25 23:25:33.886223

[debug_Tue_Mar_25_23_08_31_37720.log](https://github.com/user-attachments/files/19460460/debug_Tue_Mar_25_23_08_31_37720.log)

zhimin-zeng-bambulab, 2025-03-26:
[image]
Could you please also share the corresponding log file starting with "debug_network_"? As shown in the picture above.

michaeltavel, 2025-03-26:
I have zipped and attached all the logs.  I couldn't find that file by name anymore, I suspect the name was updated, but the data should be there.

[debug_Tue_Mar_25_23_08_31_37720.log.zip](https://github.com/user-attachments/files/19466472/debug_Tue_Mar_25_23_08_31_37720.log.zip)

michaeltavel, 2025-03-26:
An additional note, I did try resetting my printer to factory defaults.  I also formatted my microSD card and tried a different new (unused) microSD card and none of that troubleshooting changed this behavior.

kofing1, 2025-03-26:
Yes, my p1sc encountered the same issue, and after searching for an answer for two days, there are still no results.

integralmedia, 2025-03-26:
Same as issue #6182 and #6191 and #6155

jeiber, 2025-03-26:
I have a P1P upgraded to P1S and am experiencing the same behavior:

"Failed to get AMS mapping table; please select "Resume" to retry."

Postlast, 2025-03-26:
Статус: Открыто.
# 6210   Similar problem

Paliddo86, 2025-03-27:
Same on OS X version of bambustudio

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab

jeiber, 2025-03-31:
👍