bambulab/BambuStudio #6157: Unable to choose unassigned filaments in AMS

K2Kevin, 2025-03-25:
### Bambu Studio Version

2.0

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS

### Printer

BAMBU X1C AMS

### How to reproduce

I never assign filaments in any of the AMS slots.  Reset all slots, and try to print.

[image]

### Actual results

Unable to print without assigning a filament to device tab for AMS slots.

### Expected results

Should allow me to choose any slot for color selection.

### Project file & Debug log uploads

Any files, single and multicolor files are affected by this bug.  Had to roll back in order to print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ADC-Colin, 2025-03-25:
@K2Kevin is this similar to #6152 ?

fahaaaxiki, 2025-03-26:
Sorry, it is a bug, but we will fix it soon. Before the updating, you can solve it temporarily by doing these steps:
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

K2Kevin, 2025-03-26:
I will re-install latest and see it that work around works!  Thank you.

K2Kevin, 2025-03-26:
> Sorry, it is a bug, but we will fix it soon. Before the updating, you can solve it temporarily by doing these steps: Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just [...]

This works!  Thank you...

Haidiye00, 2025-03-27:
> > Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.
> 
> Asking a customer to use a different machine, with different OS is always a NO NO unless im a Beta tester. I do have Windows machines, but It's not a resolution for MAC OS version. The work around did resolve the issue on Mac OS.

Hello, we have fixed this issue with the Mac in our internal version and will release a new version as soon as possible. so sorry

K2Kevin, 2025-03-27:
> Hello, we have fixed this issue with the Mac in our internal version and will release a new version as soon as possible. so sorry

Thank you so much for the update.  I had to roll back to the previous release, the work around didnt last long before the problem started once again.  Looking forward to the MAC OS release,  my H2D arrives soon and have a few designs to be released specifically waiting for that larger bed.  Thanks again!

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.