bambulab/BambuStudio #6159: Thoughts on v2.0 Send Print Job dialog box

DARKFiB3R, 2025-03-25:
I never create time lapses with my P1S because the quality is poor, so of all the options to be displayed without clicking Advanced Options, this is the least useful to me.

I'm not sure why the "Use AMS" option exists. Either I have selected to use an external filament or I have not, so I would presume AMS useage would be automatic based on that selection?

The Auto Bed Leveling option would be my most used option, so I wish that was visable at all times, but do not understand what *Automatic Mode means. Does it mean that a quick 10s check is always performed, regardless of Auto Bed Leveling being turned on or off? Is there a setting to enable / disable this?

Thanks :)

avatorl, 2025-03-25:
The "Send print job" dialog in 2.0 is hardly usable and looks messy.

I agree that "Timelapse" is the least useful feature.

Before clicking "Send" it makes sense to quickly verify (without any clicking) if "Timelapse", "Auto Bed Leveling" and "Flow Dynamics Calibration" options are enabled or disabled and if required to enable or disable any of them with no more than one click. In 2.0 we are forced to expand "Advanced Options" just to verify status of the options. And it looks messy with a scrollbar.

[image]

Also, while a new user may need the detailed descriptions of the "Timelapse", "Auto Bed Leveling" and "Flow Dynamics Calibration" features first few prints, these descriptions quickly become useless. There is no need in always showing detailed descriptions. You can use tooltips or links to Bambu Wiki to explain the all the features for the new users.

The dialogue is used often and must be as simple as possible. Also, make the most important information visible (larger font for time and filament weight).

sylverblue, 2025-03-25:
Absolutely right Avatorl. I completely agree with you. 
I also have the problem that I can no longer select the slot of the filament I want to print with in the window.
Anybody else?

DARKFiB3R, 2025-03-25:
@sylverblue I can select the filament slot, no problem.

[image]

DARKFiB3R, 2025-03-26:
Also, who is reponsible for this image?

[image]

🤣

DARKFiB3R, 2025-04-21:
[image]

[image]

🥰

Also, thank you for the other updates included in v2.0.2 😁