bambulab/BambuStudio #6161: Keypad Enter Key Not working

Mikurie, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

1. Click on any object
2. Select Scale or S
3. Type in new scale % for X,Y, or Z. OR Type new Size using keypad on keyboard.
4. Hit Enter on far right enter on Keypad.

### Actual results

Nothing changes. If I want the new number to be entered, I have to tap the other enter key on the keyboard. The Enter key on the keypad isn't working. It works in all other programs, just not in BambuLab.

### Expected results

When I tap the Enter key, it should have updated the percent to the new number I had typed.

### Project file & Debug log uploads

[Spiral+BunnyEgg.zip](https://github.com/user-attachments/files/19456300/Spiral%2BBunnyEgg.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

andrewroth, 2025-03-28:
👍 for a fix. This is driving me a little bit crazy.

guanyun-gudujian, 2025-03-28:
fixed
will be released 
thanks

GuDuJian-J-Zhang, 2025-05-20:
@Mikurie   @andrewroth 
fixed, pls try the latest version, thanks
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_win_public-v02.00.03.54-20250424182834.exe

guanyun-gudujian, 2025-05-20:
@Mikurie @andrewroth
fixed, pls try the latest version, thanks
https://github.com/bambulab/BambuStudio/releases/download/V02.00.03.54/Bambu_Studio_win_public-v02.00.03.54-20250424182834.exe

closed 2025-10-15 by Haidiye00