bambulab/BambuStudio #6163: Unable to assign/change filament in AMS when sending job to printer

cayalag, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

masOS 15.3.2 (24D81)

### Additional system information

Since updating to Bambulab 2.0, I am unable to assign or modify which AMS slot I want to use for a particular color

### Printer

Bambu Lab P1S

### How to reproduce

To reproduce, press print Plate... in Filament section, select any of the colors included in the job. When the menu opens it shows all colors loaded in the AMS, however It doesn't allow you to modify or reassign which slot to use, as we were able to do until this version of bambustudio.

### Actual results

Unable to select desired spool.

### Expected results

It should allow you to select the slot to print from for a particular color.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sisuwest, 2025-03-25:
Glad I'm not the only one! Time to downgrade until this critical bug is fixed.

jobrun35, 2025-03-25:
Same issue. Unable to select AMS slot when printing

fahaaaxiki, 2025-03-26:
Sorry, it is a bug, but we will fix it soon. Before the updating, you can solve it temporarily by doing these steps:
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.
[image]

Haidiye00, 2025-03-26:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

sylverblue, 2025-03-27:
I don't want to blame anyone, mistakes happen. But I would be interested to know how this could happen. You test before release and yet such a trivial error occurs. I would be really interested, I am not familiar with the subject of software development. Best regards from Germany.

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00

Swift-Print, 2025-04-18:
> Hello, we have fixed this issue. Please download the latest version to use. https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50 We apologize for any inconvenience caused and will continue to improve the quality of our software.

This issue is not fixed, I am still having this issue on a windows laptop