bambulab/BambuStudio #6166: Latest Bambu Lab release doesn't include Bambu printers

lcolgantx, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1 C

### How to reproduce

Install and try to setup any Bambu printer and Bambu doesn't show in the list. I also can't add a Bambu printer manually

### Actual results

I can't use the sw because my printer isn't listed. 

### Expected results

I would like to be able to find my Bambu X1 C in the list of printers

### Project file & Debug log uploads

I can't properly setup Bambu lab after the update becuase my printer isn't listed. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lcolgantx, 2025-03-25:
I'm not using Linux, but I was able to add my printer by skipping the setup and searching for it by the IP address. I still feel this is a bug that needs to be fixed.

JohnSmithCYM, 2025-03-26:
Thanks for your feedback and we are finding the reason,you can try this:
1. Open Configuration folder by ''Help-Show Configuration Folder"
[image]
2.  Delete the 'system' and "printers" folder then restart Studio.

XunZhangBambu, 2025-03-26:
@lcolgantx Using the method provided by @JohnSmithCYM , your issue should be resolved. We currently haven’t been able to reproduce your issue. Could you please provide the log files?