bambulab/BambuStudio #6167: "Enter" doesn't work on numerical pad

InfEleXXI, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Studio

### How to reproduce

put any value in e.g. scale and please put "Enter" in num pad.

### Actual results

Doesn't work

### Expected results

Should work

### Project file & Debug log uploads

It is not necessary because it doesn't work for all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

seanbo, 2025-03-28:
I too have this issue.  I came to report the same issue.

guanyun-gudujian, 2025-04-02:
will be fixed next version

seanbo, 2025-04-16:
It works correctly for me in v2.00.02.57.  Thank you.

closed 2025-04-16 by Haidiye00