bambulab/BambuStudio #6168: 无法切换使用AMS其他料槽

chync, 2025-03-25:
### Bambu Studio Version

2.0095

### Where is the application from?

Bambu Lab Official website

### OS version

mos os 15.32

### Printer

P1S

### How to reproduce

准备打印的时候AMS料槽A1,A2均为同种材料，但是打印材料默认A1，点击无法切换到其他材料

https://github.com/user-attachments/assets/df5e1e9a-142d-46c0-b74d-cefe068b3a60

### Actual results

无法切换使用材料槽

### Expected results

点击后应该切换料槽

### Project file & Debug log uploads

[归档.zip](https://github.com/user-attachments/files/19457584/default.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fahaaaxiki, 2025-03-26:
抱歉，这是一个bug，我们会尽快修复它，在软件更新之前，您可以通过以下操作来暂时解决这个问题：
预设选择H2D机型 -> 切片并发送打印 -> 在H2D打印页面选择另一个AMS槽位(预设不匹配的警告信息是正常的，直接选AMS即可) -> 返回准备页面，选择您原本要用的机型预设 -> 切片并发送打印，此时即可正常选择AMS
如果您重启了软件，则需要重新执行以上操作，再次为引起您的使用不便致歉。

[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00