bambulab/BambuStudio #6169: H2D 0.12mm layer height profile messed up

OmiCron07, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

H2D

### How to reproduce

The 0.12mm layer height profile is messed up for H2D. The walls are halved.

[image]

### Actual results

Too thin walls.

### Expected results

Fatter walls.

### Project file & Debug log uploads

Built-in profile.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2025-03-26:
@OmiCron07 We will optimize these profiles in later versions.

OmiCron07, 2025-03-26:
@XunZhangBambu Just let you know that H2D seems 3 times slower than an X1 because of that, so it does not make good press when I compared the two printers for my existing project.

XunZhangBambu, 2025-03-26:
@OmiCron07 
Since layer heights below 0.16mm are primarily aimed at achieving higher print quality, we have directly integrated high-quality parameters for 0.12mm and 0.08mm layer heights to further enhance printing results. This effectively reduces layer lines and improves overall print quality, but it also increases printing time.

Regarding the wall thickness, the current setting is an optimized recommendation that balances excellent print quality with the best possible time efficiency. If a thicker wall is needed, it can be adjusted as required.

OmiCron07, 2025-03-26:
@XunZhangBambu Sorry, but it does not make sense. The wall thicknesses are concerned with the nozzle width, not the quality preset. You have put the wrong preset for the 0.12mm profile.

And your affirmation is incorrect because the 0.08mm profile has the right wall thicknesses:

[image]

Doubletap67, 2025-03-26:
> [@XunZhangBambu](https://github.com/XunZhangBambu) Sorry, but it does not make sense. The wall thicknesses are concerned with the nozzle width, not the quality preset. You have put the wrong preset [...]

You are 100% correct and this needs to be fixed ASAP before there's widespread use of the H2D in the community.  Right now using the 0.12 mm H2D profile instead of the 0.2 blows print time in one of my models from 4 hours to 18 hours, and only increases print time on the X1C from 4 hours to 5 hours.

james7360, 2025-03-27:
@XunZhangBambu This isn’t about optimization—it’s about a clear mistake. The 0.12mm profile is using settings meant for a 0.2mm nozzle, even though the printer has a 0.4mm nozzle.

You could leave it as-is, but then videos like this will keep popping up, showing the H2D running 3x slower than it should: https://www.youtube.com/shorts/7hiyJ62Ntdk

I’ve already fixed it for myself, so it doesn’t affect me directly. But for new users, this makes your flagship printer look unnecessarily slow.

XunZhangBambu, 2025-03-28:
@OmiCron07 @james7360 @Doubletap67 
Sorry for the initial misunderstanding regarding your feedback about wall layers. We've confirmed the line width was mistakenly set for a 0.2mm nozzle configuration, we will fix this asap.

XunZhangBambu, 2025-03-31:
@OmiCron07  This issue has been resolved in the latest release(2.0.1.50). Thanks for your report

closed 2025-03-31 by XunZhangBambu

james7360, 2025-04-01:
> [@OmiCron07](https://github.com/OmiCron07) This issue has been resolved in the latest release(2.0.1.50). Thanks for your report

The 0.12mm fine profile still defaults to 4 walls when using a 0.4mm nozzle. I don't think that's intentional, since all the other presets including the 0.08mm profile use just 2 walls. 4 walls is the default settings for the 0.2mm nozzle.

ffafe5, 2025-04-02:
> The 0.12mm fine profile still defaults to 4 walls when using a 0.4mm nozzle. I don't think that's intentional, since all the other presets including the 0.08mm profile use just 2 walls. 4 walls is [...]

Created new issue since this one was closed prematurely: https://github.com/bambulab/BambuStudio/issues/6388