bambulab/BambuStudio #6170: Random error when connecting with upgraded P1P

xKamalG, 2025-03-25:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P with P1S conversion

### How to reproduce

1. Go to device tab
2. Select upgraded P1P

### Actual results

Popup window with 'Unknown Error 0500_C011 235825'

[image]

### Expected results

No error

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-04-10:
@xKamalG The error does not matter, you can ignore it. We will fix the popup later

BoBoYu-BamBu, 2025-04-23:
@xKamalG Sorry, could you please provide all the files in the configuration folder? Please refer to the screenshot below to package all the files. This will facilitate our analysis and problem-solving.

[image]
[image]

XinZhangBBL, 2025-04-25:
Halo, The problem solved or not ? If there is no reply, we will close this issue。

XinZhangBBL, 2025-05-19:
You can reopen the issue if not resolved