bambulab/BambuStudio #6171: 2.0.0 Completely broken on ARM64

SalmonSoup15, 2025-03-25:
### Bambu Studio Version

2.0.0

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 24H2 26100.3613 ARM64

### Additional system information

1.x although slow, worked perfectly fine on ARM64 (specifically Snapdragon X Elite x1e-80-100) however with the 2.0.0 release, it simply says "loading configuration" for about 2 seconds and then closes. Nothing shows in task manager. 

### Printer

P1S AMS Combo

### How to reproduce

1. Download and install on an ARM64 machine
2. Open Bambu Studio

### Actual results

Shows loading screen for about 2 seconds, then closes. nothing shown in task manager.

### Expected results

Bambu studio opens and works normally

### Project file & Debug log uploads

Cant upload any relevant files as it never opens in the first place

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-03-26:
currently it is not supported on windows arm64
and we are trying to build an independent version

please just wait our finish

begratefull, 2025-03-26:
Can confirm. Bambu Studio hasn't worked on ARM64 since 1.10, it worked before then... Guess I'm sticking with 1.09 until you finish supporting arm64.

SoCuul, 2025-03-26:
> Can confirm. Bambu Studio hasn't worked on ARM64 since 1.10, it worked before then... Guess I'm sticking with 1.09 until you finish supporting arm64.

Is there a reason you can't use the x64 build with window's built in x86 emulation for arm?

SalmonSoup15, 2025-03-26:
> Is there a reason you can't use the x64 build with window's built in x86 emulation for arm?
Thats what we're trying, it just doesn't work with 1.10 and above

RyanEwen, 2025-03-29:
OrcaSlicer still works fortunately, for what that's worth

konzolec, 2025-03-29:
Orca is the way then.

Goodwine, 2025-04-02:
duplicate of https://github.com/bambulab/BambuStudio/issues/6014

Goodwine, 2025-04-02:
> OrcaSlicer still works fortunately, for what that's worth

not if the printer has the new firmware that blocks unauthenticated MQTT packets

RyanEwen, 2025-04-02:
> > OrcaSlicer still works fortunately, for what that's worth
> 
> not if the printer has the new firmware that blocks unauthenticated MQTT packets

Considering Bambu won't even launch, I can at least use Orca to slice a file and get it onto the printer in other ways.

Probably worth mentioning, Bambu Connect doesn't launch on ARM either.

lanewei120, 2025-04-07:
please check https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560
for the current solution

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120