bambulab/BambuStudio #6173: Bridging on top surface with solid surface below

chaosclarity84, 2025-03-26:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C and P1S

### How to reproduce

Slice a part made in Fusion 360 with threaded holes.

### Actual results

Around any threaded hole, Bambu Studio puts a bridge patch around it. The more walls I add, the bigger the bridge patch is. Several versions back, this did not happen.

### Expected results

There should be no bridges on top of a solid surface underneath.

### Project file & Debug log uploads

[Example_Project.zip](https://github.com/user-attachments/files/19458918/Example_Project.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-26:
Hello, please update to the latest official version V2.0.0.95 on our official website and give it a try.
Download Address：https://bambulab.com/zh/download/studio

chaosclarity84, 2025-03-26:
> Hello, please update to the latest official version V2.0.0.95 on our official website and give it a try. Download Address：https://bambulab.com/zh/download/studio

I also just tried that hoping it would help - but did not. Did you check the project file?

anlinas, 2025-08-21:
Experiencing the same issue

**Bambu Studio Version**
2.02.00.85

**Where is the application from?**
Bambu Lab Official website

**OS version**
Windows 11

**Printer**
P1S, 0.4mm nozzle

**How to reproduce**
Slice a part with threaded holes. Not sure if it was originally modelled in Fusion 360. Using higher layer heights:
0.4mm nozzle
- 0.20mm
- 0.24mm
- 0.28mm

0.2mm nozzle 
- 0.10mm
- 0.12mm
- 0.14mm

Bug does not appear when slicing with lower layer heights. 

**Actual results**
Around any threaded hole, Bambu Studio uses bridging on the top surface around about a third of the circumference of the hole. 

**Expected results**
There should be no bridges on top of a solid surface underneath.

**Project file & Debug log uploads**
https://makerworld.com/en/models/671539-coffee-mug-holder-v-2#profileId-1684521