bambulab/BambuStudio #6174: Timelaps option is gone from Send print Job

whalesolution, 2025-03-26:
### Bambu Studio Version

2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S

### How to reproduce

When you want to print a new object there is no toggle anymore to turn timelaps on or off

### Actual results

No timelaps toggle

### Expected results

Bring back the timelaps toggle

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fahaaaxiki, 2025-03-26:
May I ask if there is an SD card inserted into your printer? If you haven't inserted the SD card, the option for Timelapse won't be displayed in the new version.

fahaaaxiki, 2025-03-26:
Sorry, I cannot reproduce your problem here. Timelaps toggle displayed normally in my print job.

[image]

Could you please provide your log file? We will look into this issue, thanks

Bluedai, 2025-04-07:
On the first day after the update, I had the same issue — the slider for Timelapse was missing. Below the filament section, there was only the 'Advanced Options' button, and behind that, just three settings. Even though I didn’t make another software update, today on the second day, the Timelapse option is suddenly visible directly on the 'Send Print Job' page. I have an X1C.

Bluedai, 2025-04-07:
Here are my log files from the day I had the same problem

[log.zip](https://github.com/user-attachments/files/19629106/log.zip)

XinZhangBBL, 2025-04-16:
@whalesolution @Bluedai @jwise the problem has been fixed. https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

closed 2025-04-16 by XinZhangBBL