bambulab/BambuStudio #6175: O Ícone Salvar continua mostrando como "não salvo"

GuilhermeParizotto, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab X1c

### How to reproduce

File -> Save Project está desabilitado
Botão Salvar continua habilitado
Quando o projeto foi salvo, sem alteração, o botão salvar deveria ser exibido na cor cinza e sem ação (Semelhante aos botões Avançar e Retroceder)
[image]

### Actual results

Atualmente, ao salvar, o menu File -> Save Project fica na cor cinza e desabilitado
Porém, o botão Salvar continua na cor branca, demonstrando que o projeto não foi salvo

### Expected results

O correto, seguindo o padrão dos softwares, é que o botão salvar troque de cor seguindo o que acontece com o menu File -> Save Project

### Project file & Debug log uploads

File -> Save Project
Botão Salvar

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-03-26:
Could you offer a detail or a screenshot to show your question,I have tried at least two way and it all worked:
- Import a project and change anything in it(including process params change/move model...)
- Import a stl to an empty project.

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

closed 2025-04-16 by Haidiye00

GuilhermeParizotto, 2025-04-16:
Olá, tudo bem? Obrigado pelo seu interesse em responder.
Atualizei o Bambu Studio para a última versão citada e continua funcionando da mesma forma.
O que eu quero dizer é que o ícone salvar nunca fica desabilitado. Sabe quando um botão é "não clicável" que ele fica em uma cor cinza? Pois é, o ícone Salvar nunca fica desabilitado.
Eu fui programador por 6 anos, e isso acaba me incomodando toda vez que preciso abrir o menu para ver se o projeto está salvo ou não, por que o ícone Salvar fica sempre me mostrando que o projeto não está salvo, mesmo quando está.
Se você deseja simular o que eu falo, abra um projeto, salve o projeto, mande o print aqui. O Ícone do Salvar fica desabilitado em algum momento? No meu Bambu Studio nunca fica. E deveria ter o mesmo comportamento do menu ("Save Project Ctrl+S")

[image]

GuilhermeParizotto, 2025-04-16:
Note a flexa para a direita (->) na primeira imagem ela está na cor branca (habilitada), na segunda imagem está na cor cinza (desabilitada). Esse comportamento que o ícone Salvar deveria ter. Consegui explicar?

Haidiye00, 2025-04-17:
[image]
Hello, I didn't quite understand what you meant. After testing and saving, I have disabled the save button.

GuilhermeParizotto, 2025-04-17:
Aqui pra mim fica sempre habilitado o botão salvar. Segue um pequeno vídeo demonstrando
https://github.com/user-attachments/assets/1f7760de-5dbe-4830-90ba-6e990e13a4a5

Haidiye00, 2025-04-18:
Hello, may I ask if you are using the latest version? We have tested and found that the latest version of Windows is working properly.

GuilhermeParizotto, 2025-04-19:
Sim, estou utilizando a última versão, na versão anterior também acontecia

[image]

Haidiye00, 2025-04-21:
Hello, this is a bit confusing for me~