bambulab/BambuStudio #6176: BAmbu STdio Crashed when I slice a plate

maskinazi, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4  Beta (24E5238a)

### Printer

Bambu Lab PS1

### How to reproduce

Slice Plate 7

### Actual results

Bambu Studio Crashes

### Expected results

I am unable to upward the 3mf file.  I keep getting the message 

### Project file & Debug log uploads

[debug_Tue_Mar_25_22_55_40_35211.log.0.zip](https://github.com/user-attachments/files/19459875/debug_Tue_Mar_25_22_55_40_35211.log.0.zip2>

### Checklist of files to include

- [x] Log file
- [x] Project file

maskinazi, 2025-03-26:
I am unable to upload the 3mf file.  I keep getting an error message when I try to upload it.

BoBoYu-1, 2025-03-26:
Hello, could you please try to provide us with the 3mf file and the crash log?
you can find them following this guide
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
and please send as many as you can of these crash reports

maskinazi, 2025-03-26:
Ball JointedmDoll3.3mf
<https://drive.google.com/file/d/1W_f_R6y86fiFqXJJY9DkBMi8-Ejih_vk/view?usp=drive_web>
I wasn't able to upload the 3mf file to the crash report.

QingZhangBambu, 2025-05-23:
@maskinazi plate 7 be sliced successfully on V2.0.3.54. Could you try it?

[image]