bambulab/BambuStudio #6178: [Linux] AppImage fails on Ubuntu 24.04 / Mint Noble due to outdated WebKitGTK dependencies

longrackslabs, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 22 Cinnamon

### Additional system information

System:
  Kernel: 6.8.0-55-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: 1: GDM3 v: 46.2
    2: LightDM v: 1.30.0 Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    uuid: <superuser required> UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 55% (should be ignored) rechargeable: yes status: discharging
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3+ rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 3146 high: 4725 min/max: 2200/4850 boost: enabled cores: 1: 3648 2: 2200
    3: 2200 4: 4725 5: 3492 6: 2200 7: 2200 8: 2200 9: 2200 10: 2200 11: 3780 12: 3731 13: 3774
    14: 4725 15: 3823 16: 3248 bogomips: 121589
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: nvidia v: 535.183.01 arch: Maxwell
    pcie: speed: 8 GT/s lanes: 16 ports: active: none off: HDMI-A-1 empty: DP-1,DVI-D-1,DVI-I-1
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: nouveau
    unloaded: fbdev,modesetting,vesa failed: nvidia gpu: nvidia,nvidia-nvswitch display-ID: :0
    screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 122 s-size: 533x302mm (20.98x11.89") s-diag: 613mm (24.12")
  Monitor-1: HDMI-0 res: 2560x1440 hz: 60 dpi: 124 size: 526x296mm (20.71x11.65")
    diag: 604mm (23.76") modes: N/A
  API: EGL v: 1.5 hw: drv: nvidia platforms: device: 0 drv: nvidia device: 2 drv: swrast
    surfaceless: drv: nvidia x11: drv: nvidia inactive: gbm,wayland,device-1
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 535.183.01 glx-v: 1.4
    direct-render: yes renderer: NVIDIA GeForce GTX 950/PCIe/SSE2
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  API: ALSA v: k6.8.0-55-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    RTL8111/8168/8411 driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000
    bus-ID: 2a:00.0 chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
  IF-ID-1: docker0 state: down mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 driver: btusb v: 0.8 type: USB rev: 2.0
    speed: 12 Mb/s lanes: 1 bus-ID: 1-6:4 chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 4.0 lmp-v: 6 sub-v: 220e
    hci-v: 6 rev: 1000 class-ID: 7c0104
Drives:
  Local Storage: total: 1.95 TiB used: 696.9 GiB (35.0%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 tech: SSD serial: <filter> fw-rev: 111130WD temp: 42.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
[... 126 lines cut ...]

These libraries are no longer available in current package repositories.

The user must create symlinks to newer libraries (4.1) to get the AppImage working

### Expected results

AppImage should launch cleanly on modern Linux distributions without requiring manual library symlinks.

Either:

Dependencies like WebKitGTK are bundled inside the AppImage, or

The AppImage is rebuilt against currently supported library versions (e.g., GTK 4.1 / WebKitGTK 4.1), or

An official Flatpak version is provided for better compatibility across distros

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-03-26:
[image]

pfeerick, 2025-03-29:
Why are you running Bambu_Studio_ubuntu-20.04_PR-6155.AppImage on a ubuntu 24.04 system? try running the 24.04 AppImage instead... it is working just fine for me on Mint 22.1

longrackslabs, 2025-03-29:
Oh wow—yep, that one’s on me. I didn’t realize there were separate AppImages for each version. Totally missed it. With dyslexia, version numbers and filenames don’t always register the way they should. It’s one of those things that slips past until something breaks.

pfeerick, 2025-03-29:
> Sounds like we can close the file on this one.

lol... know what you mean! You should be able to close the issue if that is
all it was, that's the other option at the bottom of the form. While it can
be confusing, I'm glad there was some thought into keeping both an older
version and newer version of the AppImage, just in case you are still on an
older LTS system, etc.

closed 2025-03-29 by longrackslabs