bambulab/BambuStudio #6179: update messed up my printer

dtf5629, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 pro

### Additional system information

System Model	OptiPlex 7050
Processor	Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz, 3600 Mhz, 4 Core(s), 8 Logical Processor(s)
OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
Total Physical Memory	31.8 GB

EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

### Printer

Bambu A1

### How to reproduce

EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

### Actual results

EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

### Expected results

EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

### Project file & Debug log uploads

EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Sirrux0815, 2025-03-26:
I have the same issue. AMS unplugged and then it went on

BoBoYu-1, 2025-03-26:
Thank you for your feedback. Could you record a screen shot or take a screenshot to describe your problem? This would be helpful for us to reproduce the issue and solve it.

JohnSmithCYM, 2025-03-26:
We have found that it may be caused by "Use AMS" was abled when send print for single nozzle. Disbale it when sending a print and it will use external spool to print.
[image]
We are working on this bug and solve this.

Sirrux0815, 2025-03-26:
Oh, I overlooked the fact that there is another menu there. But if you use the external spool, shouldn't it automatically be clear that the AMS doesn't need to be activated?

JohnSmithCYM, 2025-03-26:
@Sirrux0815 We know this could mislead a lot of people and we will try to fix it soon.Thanks again.

dtf5629, 2025-03-26:
EXT SPOOL WON'T WORK WITH AMS PLUGGED IN AFTER UPDATE....

TURNING AMS OFF IN MENU DOESN'T WORK EITHER.
I HAVE BEEN SO HAPPY WITH THIS PRINTER AND SOFTWARE TILL NOW.. 
MAKING SOFTWARE PRETTY MAY BE NICE, BUT I WOULD LIKE IT NOT TO MAKE MY LIFE HARDER..
I HOPE IT GETS FIXED SOON..
THANK YOU IN ADVANCE FOR YOUR HELP....