bambulab/BambuStudio #6181: P1P -> P1S Upgrade Kit | Cannot sync AMS filaments

AustinAHunter, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P, upgraded to P1S

### How to reproduce

1. Ensure that the P1P has the upgrade kit enabled on its screen.
2. Go to the device tab in Bambu Studio
3. Ensure that your P1P (P1S) printer is selected
4. Go to the Prepare tab and attempt to sync filaments

### Actual results

Error popup

[image]

### Expected results

Filaments are synced from the AMS to the project

### Project file & Debug log uploads

[mini-rack.zip](https://github.com/user-attachments/files/19461207/mini-rack.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-03-26:
Hello, we will allow AMS synchronization based on compatible models in the future. Thank you for your feedback~

AustinAHunter, 2025-03-26:
I believe this was a breaking change from the prior version of Bambu Studio. Are you meaning to say that this bug will be fixed or was this an intentional change to the way that this feature used to work?

Haidiye00, 2025-03-26:
Hello, we will optimize this behavior. That is to say, compatible models can also perform AMS synchronization and initiate printing.

IOIO72, 2025-03-26:
> Hello, we will optimize this behavior. That is to say, compatible models can also perform AMS synchronization and initiate printing.

My understanding of your answer is as follows:

I need to do the following steps:
1. Change the settings in Bambu Studio for my upgraded P1S to P1P
2. Synchronize the AMS 
3. Switch settings back to P1S.

Is this what you suggest? I hope this bug is fixed in the next release so I don't have to switch back and forth.

Haidiye00, 2025-03-27:
Hello, p1s and p1p are compatible machines. If your preset is p1s and the connected printer is p1p, both can initiate AMS mapping and printing tasks.
We will fix it in subsequent versions~

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00