bambulab/BambuStudio #6183: cant login bambu studio . page is blank

Niv1214, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

x1c

### How to reproduce

go to log in 

### Actual results

blank page

### Expected results

open login page

### Project file & Debug log uploads

no 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-26:
Thank you for your feedback. Could you please let me know if you are in a domestic environment or an overseas environment? Also, is your computer's network connection working properly?

Niv1214, 2025-03-26:
> Thank you for your feedback. Could you please let me know if you are in a domestic environment or an overseas environment? Also, is your computer's network connection working properly?

I am in a domestic environment and my computer's network connection working properly. [image]

BoBoYu-1, 2025-03-26:
We just tested the login process within the domestic environment, and the login window loads within about 5 seconds. It seems this is most likely a network connectivity issue on your computer. Could you try switching to a different network connection?

Niv1214, 2025-03-26:
> We just tested the login process within the domestic environment, and the login window loads within about 5 seconds. It seems this is most likely a network connectivity issue on your computer. Could you try switching to a different network connection?

I tried 2 different networks and still nothing.

ahtinurme, 2025-03-30:
You need to go the Preferences and select languages and then login appears.

vkukk, 2025-04-04:
> You need to go the Preferences and select language and then login appears.

I can verify that this workaround is correct!

realzorro, 2025-04-16:
Please choose language under Bambu Studio settings, this issue is resolved.