bambulab/BambuStudio #6185: Can't choose filament

PandaBamboozle, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.3.2 (24D81)

### Printer

X1 carbon

### How to reproduce

Choose print in app and in the popup window before i could choose one of the AMS filaments, now it still says I can choose, however when I click on them nothing is happening.

### Actual results

Cannot choose on of the 4 AMS filaments I want to use, it shows me the filaments but does not let me choose. I click and nothing happens.

### Expected results

Should let me choose the needed filament. 

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-26:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

If you continue to use a Mac laptop, you can avoid this problem by following these steps：
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

tabl5, 2025-03-26:
Same issue here MacOS, the previous version was able to change the color after the slice before sending to the printer but now that is broken..  my work around was to go and select the color in the objects before slicing.

samalfano, 2025-03-26:
I'm having the exact problem on the Mac version of 2.0.

Haidiye00, 2025-03-27:
Hello，We have optimized the behavior in the internal version：p1s and p1p are compatible machines， If your preset is p1s and the connected printer is p1p, both can initiate AMS mapping and printing tasks.
We have fixed it and will release a new version as soon as possible~