bambulab/BambuStudio #6186: Takes 10x longer to Slice plate

PandaBamboozle, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.3.2 (24D81)

### Printer

X1 Carbon

### How to reproduce

I'm printing a 3d map in the size of 23x23x5 cm. I did it multiple times before with previous versions of software and everything was quick and easy - less than half a minute to slice the project. Now after the update it takes around 5 minutes to slice the object for print.

### Actual results

Sometimes gets stuck, sometimes slices for over 5 minutes for a 5 cm high object.

### Expected results

Was much faster before, should be much faster now.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-03-26:
Thanks for your feedback and we want to know more about your problem, can you offer an example 3mf file for us to test? You can upload a .zip here.

PandaBamboozle, 2025-03-26:
Here you go. 
[

[Venecija.3mf.zip](https://github.com/user-attachments/files/19468521/Venecija.3mf.zip)

](url)

XunZhangBambu, 2025-03-31:
@PandaBamboozle This issue has been resolved in the latest release(2.0.1.50). Thanks for your report

closed 2025-03-31 by XunZhangBambu