bambulab/BambuStudio #6187: Only single height range available "Height range modifier".

shodan65, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S 

### How to reproduce

1. Select Object
2. In context window pick "Height range modifier"
3. Try to add a second range of layers in the advanced menu on the left hand side

### Actual results

Only a singe range is available, no way to add another one or even better multiple other ranges.

### Expected results

I would like to be able to add and maintain multiple ranges. A duplicate function would also be handy.

### Project file & Debug log uploads

[FatCat.zip](https://github.com/user-attachments/files/19466805/FatCat.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-1, 2025-03-27:
Hello, multiple Height Range Modifiers can be added using these steps:

Select the object, then right-click to add a Height Range Modifier;
Under the "Process - Object List" below, select the "Layer" tab, where you can add multiple Height Range Modifiers in the lower-right corner of this page;
As demonstrated in the GIF video below:
[image]

shodan65, 2025-03-27:
Nice tey, but there ist No plus sign in mm y Version of the Studio.

Haidiye00, 2025-03-28:
Hello, can you take a screenshot of your software and see why there is no+sign

shodan65, 2025-03-28:
Sorry, forget it. I'm blind. Basically I

closed 2025-03-28 by shodan65