bambulab/BambuStudio #6188: 软件崩溃

smallmj, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

用新版软件打开这个模型：https://makerworld.com.cn/zh/models/1068221-tuo-zhu-liao-pan-zhi-jia#profileId-1109142，默认选择H2D，手动切换到P1S时，Bambu Studio直接崩溃闪退

### Printer

P1S

### How to reproduce

1。下载https://makerworld.com.cn/zh/models/1068221-tuo-zhu-liao-pan-zhi-jia#profileId-1109142并用新版软件打开
2. 手动切换到P1S

### Actual results

Banbu Studio崩溃闪退

### Expected results

成功切换到P1S

### Project file & Debug log uploads

[Spool+Holder+by+Bambu (1).zip](https://github.com/user-attachments/files/19468050/Spool%2BHolder%2Bby%2BBambu.1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-03-26:
Thanks for your feedback and we have confirmed it crash when change preset.We will try to fix it soon.

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab