bambulab/BambuStudio #6192: Blank Login Page

Orel-A, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 24H2 Hebrew

### Printer

Bambu Lab A1

### How to reproduce

Use a Windows language that doesn't exist on Bambu Studio

### Actual results

The login page will be blank because the Language isn't set in Bambu Studio.
Settings -> Preferences -> Language is empty!

### Expected results

If the Language isn't determined from the Windows default language it should be set to English!

### Project file & Debug log uploads

N\A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Orel-A, 2025-03-30:
> You need to go the Preferences and select languages and then login appears.

Yes, this is indeed the solution. However, the issue was filed so Bambu can fix it in their code in the first place. If you search this issue on Google you will find people telling you to delete Edge or other Windows files, corrupting your Windows installation.

realzorro, 2025-04-07:
Please go to Bambu's official website and try downloading and installing the latest version of BambuStudio. We hope it can be helpful for your problem.

Orel-A, 2025-04-07:
> Please go to Bambu's official website and try downloading and installing the latest version of BambuStudio. We hope it can be helpful for your problem.

Doesn't work! Go to %AppData%/BambuStudio.conf file, change language to `"language": "he_IL"` you will see my problem. Login page is blank.