bambulab/BambuStudio #6193: Not lunching because of missing en_GB locales

Twikzer, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux

### Additional system information

kernel 6.13.8, KDE plasma 6.3.3, wayland

### Printer

not aplicable

### How to reproduce

1. Install it from aur / download the appimages from github (same resuld with the 3 of them)
2. launch the program

### Actual results

you get a message that tells you to add en_GB as a language in your computer
[image]

### Expected results

The program launch normaly.

### Project file & Debug log uploads

not aplicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Twikzer, 2025-03-26:
A temporal solution is to:
- Do as the message tells you (i do it with en_CA also, and read online that en_US also works)
- edit ~/.config/BambuStudio/BambuNetworkEngine.conf and change en_US to you language
- remove the en_GB (or orther en_XX) from your system

and the program lauch normaly.

This does NOT change the program language, the program will be in english until you change it inside the program, this is expected behavior. This is only for the program to launch in non-english computer.

MackBambu, 2025-03-28:
I’ve traced through the code and temporarily removed the consistency check between the system language and the application language on Linux. I’ve also disabled the fallback to en_GB when no supported language is found.
Please test and let me know if the issue still persists.
ubuntu 20.04: https://github.com/MackBambu/BambuStudio/actions/runs/14108557106/artifacts/2832541275
ubuntu 24.04: https://github.com/MackBambu/BambuStudio/actions/runs/14108557106/artifacts/2832446726

MackBambu, 2025-03-28:
trace https://github.com/SoftFever/OrcaSlicer/issues/7649

MackBambu, 2025-03-28:
I'm a bit confused about whether software on Linux needs system language support to set its own language.
If it does require support, then we should revert to the original Prusa code, where switching the language checks if the language is supported, and shows a dialog box if it's not.
If it does not require** the system language and the application's language to match**, then we don't need to perform this check at all.

Twikzer, 2025-03-29:
> I’ve traced through the code and temporarily removed the consistency check between the system language and the application language on Linux. I’ve also disabled the fallback to en_GB when no [...]

i try both versions, and they work identically. it lunch with a black square, and then it closes, when you launch it again, it launch normally.

I delete ~/.config/BambuStudio between test.

This versions create "ota", "printers", "system", "user" folders and "BambuStudio.conf", besides the "cache" and "log" folders that the current version creates. (inside ~/.config/BambuStudio)

Twikzer, 2025-03-31:
No, i have es_ES (spain spanish)

with the new appimage (the ones you publish in this issue), the en_GB problem does not exist, the problem is with the old (main branch) appimage.

The appimage on main branch shows the en_GB requirement, and does NOT generate the BambuStudio.conf file if you dont have some english language installed on your computer.

I use the canada version because i had en_US (and sometimes en_GB) overwrite some es_ES configs (on windows, debian and popOS) so i decided to try with other english country

The new appimage generates the BambuStudio.conf with en_US, but the program lunches anyways, even if i only have es_ES installed. I try lunching the main branch appimage to check if i still have some english language installed, and it give me the error.

ZefengWang, 2025-04-03:
why last issue closed?
it didn't fixed yet, it has problem with language detection
you may check what env variable it read, check it if correct, and check if the appimg is same as run application directly. it's import for last tip.

ZefengWang, 2025-04-03:
there are LANGUAGE and LANG two env variables, which one you have read?

ZefengWang, 2025-04-03:
you may ref this web site
https://superuser.com/questions/392439/lang-and-language-environment-variable-in-debian-based-systems

jrtorrez31337, 2026-05-09:
Confirming this is still hitting users on **v02.06.01.55** (the latest 2.6.1 public-beta build) on Ubuntu 24.04 today (2026-05-09). The `locale-gen` workaround in the comments above still applies and still works.

Two observations that may help maintainers prioritize, since this thread is approaching its 14-month anniversary:

**1. The failure mode is silent on a non-trivial fraction of Linux setups.** The "Switching language failed" popup window can launch behind other windows and never get focused — from the user's perspective the app just "doesn't open." On my system the AppImage's bash wrapper (`/tmp/.mount_BambuS<random>/bin/bambu-studio`) keeps running even after the actual GUI binary `bambustu_main` exits, so users running naive `pgrep bambu-studio` see a "running" process and conclude their window manager is broken rather than the app. I burned a couple of hours before finding the popup. A first-launch failure that takes hours to diagnose is a rough first impression for new Linux users.

**2. There's a working patch in this thread already.** @MackBambu posted patched Ubuntu 20.04 + 24.04 builds on 2025-03-28 that "removed the consistency check between the system language and the application language on Linux" and "disabled the fallback to en_GB when no supported language is found." That's exactly the right fix. Five-plus minor versions have shipped since without merging it. Is there a blocker we don't know about, or is this just stuck waiting for a maintainer pass?

Happy to test a candidate PR if one materializes.