bambulab/BambuStudio #6197: Bambu Studio 2.0 does not pick up temperatures after resyncing the filaments

stegula, 2025-03-26:
### Bambu Studio Version

2.0

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

Bambu Studio after the update does not update the Bed Temperature upon "resyncing" the filament. It stays on the previous value - which in my occasion is TPU - Bed temp 35C, when changed to PETG, the bed temp stays 35C

### Printer

P1S no AMS

### How to reproduce

1. Open BS and memorize the currently selected filament and its temperatures
2. Insert new model
3. Change filament and memorize its temperatures
4. Slice
5. Print
6. Observe the temperatures

### Actual results

The temperatures do not change - possible the material algorytm does not as well

### Expected results

The material to change

### Project file & Debug log uploads

This is not entirely aplicable by uploading these.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ravenge-au, 2025-03-27:
I also have been having adhesion issues with AMS, Textured PEI plate and P1s after updating to 2.0.  80+ prints with zero failures. first 3 prints on 2.0 had adhesion issues. First party Bambu PLA filaments only.

alfrankgit, 2025-03-29:
I can confirm this bug.

I just started the first print on my A1 mini with Bambu Studio 2.0.1.50 and saw that the build plate temperature was set to 65 °C even though my own filament profile was selected where the build plate temperature for Textured PEI is at 50 °C.

The wrong value seems to come from the default filament profile "Bambu PLA Basic".

AlexHolly, 2025-04-17:
[image]

JohnSmithCYM, 2025-04-17:
Similar to what mentioned in #6565

JohnSmithCYM, 2025-04-17:
@AlexHolly thanks for your feedback, but as I mentioned down, I can't reprouduce your fault.I believe it should be a bug, I want to know more step about your opeartion.
- Can you reproduce it directly? I mean, you just open the Studio,and slice,then it is wrong, right?
- Can you offer your 3mf for us to have a try?
- Some people said it happened after change filament and the other said when you quit Studio and open it again, the qeusiotn disappear, did that work for you ?
-

[image]

AlexHolly, 2025-04-17:
It works when I open the 3mf file.

It is not working when I open a Model from "Online Models" then it opens everything with different settings "Bambu Lab .." Printer instead of my "Anycubic ..." I have to switch filament Settings from this:

[image]

When I slice with my settings it will use incorrect temp.

JohnSmithCYM, 2025-04-18:
[image]
@AlexHolly I followed the whole process as what you reported and it still works with the right temperature(you can see the gif above).I am so confused that how can I reproduce this problem, it must be something was changed and my environment can't reproduce that.
I need more to give it a test:
- Can you give a whole gif or video to show the whole steps from open studio to the slice output.
- Can you upload your all config file(like what in the red frame below). You can zip it and send to malkey.chen@bambulab.com, we make sure no data leak and the only use is to find out this annoying problem.
[image]

AlexHolly, 2025-04-18:
1. Open Application
2. Select a Model from "Online Models" for example "Mold for Christmas Ball Candle"
3. "Download and Open"

https://github.com/user-attachments/assets/b2853d2c-845e-4e22-9a5e-9dcb99213b0f

Check email for logs and config.

JohnSmithCYM, 2025-04-20:
@AlexHolly Thanks for your feedback and I have reproduced your problem, we are finding why this happened. Thank again.

JohnSmithCYM, 2025-04-22:
@stegula @ravenge-au @alfrankgit @pr3sidentspence We want to make sure your problem is same as AlexHolly, could anyone provide a wrong gcode file here and tell us your profile when slicing.

pr3sidentspence, 2025-04-23:
I  don't get a wrong code, it's just the gcode after slicing has 35C for
the bed temp.

JohnSmithCYM, 2025-04-23:
@pr3sidentspence Can you reproduce this problem steady now?

pr3sidentspence, 2025-04-23:
I haven't tried printing with my Ender 3 in a bit. I will soon and let you
know.

JohnSmithCYM, 2025-04-23:
Okay, your problem is the same with AlexHolly due to 3rd-party printer's hot bed reason, it is confirmed.

JohnSmithCYM, 2025-04-24:
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54 We have solve the Bambu Printer‘s question with the new verisoin, please the poster with X/P/A printer give it a try.

@pr3sidentspence @AlexHolly The 3rd-party printer (like ender-3 and Anycubic Vyper) still need some time to fix its problem, we have found at least two issue with this problem. We will let you know it when we are ready.

BuongiornoTexas, 2025-11-10:
Just ran into this (I think?). I started with a 3 plate model with TPU and PLA elements. I deleted both PLA plates, and the PLA filaments, set up my preferred TPU profile, and the print ran at a bed temperature of 70C rather than 35C specified in the TPU preset. 

To make things even stranger, none of the bed temperature settings in the model were above 55C, and all of my recent printing has been at 35C. 

At very least, BambuStudio sent the wrong temperature to the machine. Unfortunately, I'm not sure I can repeat this. 

P1S with AMS2, feeding from an external spool.

When I picked this up, I stopped the print, exited BambuStudio, and restarted the print. The next piece is printing as expected (and much higher quality).