bambulab/BambuStudio #6202: Can't select the filament when sync with AMS and when print

Ddcr13, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sequoia 15.3.2 

### Additional system information

CPU: Apple M1
Memory 16GB

### Printer

Bambu Lab P1S

### How to reproduce

1 - Go to "Sincronize filament list from AMS" in Project Filament tab
2- Try to Map one filament.
 
Or

1- Go to "Print Plate"
2 - Click on the box to select the filament.
3 - Try to select the chosen filament.

### Actual results

Can't select the filament.

[image]

### Expected results

Select the filament manually

[image]

### Project file & Debug log uploads

Doesn't have a log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

onurdemir, 2025-03-26:
I was about to create the exact same ticket. This is so annoying.

emjprints, 2025-03-27:
I'm having the same issue.

Haidiye00, 2025-03-27:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

If you continue to use a Mac laptop, you can avoid this problem by following these steps：
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00