bambulab/BambuStudio #6204: P1P with official enclosure kit

Grinch74, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

 

### Printer

P1P with official enclosure kit

### How to reproduce

1. Prepare
2.  choose the P1S Profile (cause i have the enclosure kit)
3. try to scnc from AMS
4. Scree: Thecurently conected Printer .. in not an P1S please switsch to an P1S
5. i cant switsch to an P1S cause the device was origianly an P1P but with the official enclosure kit its now a P1S with all funcionalty
6. i chose a P1P again
7. sync with the p1p 
8. The ams sync now 
9. slice 
10. try to print with the p1p setting 
11. new window opens: The printer type selected when generating G-Code is not consitent with the currently selected printer .... and so on 
12. i can confirm or cancel
13. when i go back and just change the profile to an P1S all go fine without any error window

### Actual results

in both ways i can print but this shouldnt happen with the official enclosure kit it should recognise the P1P as an P1S why else i can select this official enclosure kit on the P1P menu directly on the printer menu

### Expected results

that i can use the P1S profile with the P1P and the official enclosure kit without any problems  

### Project file & Debug log uploads

[debug_Wed_Mar_26_17_37_04_31472.log.zip](https://github.com/user-attachments/files/19474422/debug_Wed_Mar_26_17_37_04_31472.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-27:
Hello, p1s and p1p are compatible machines. If your preset is p1s and the connected printer is p1p, both can initiate AMS mapping and printing tasks.
We will fix it in subsequent versions~

Grinch74, 2025-03-27:
> Hello, p1s and p1p are compatible machines. If your preset is p1s and the connected printer is p1p, both can initiate AMS mapping and printing tasks. We will fix it in subsequent versions

Yeah thx. I know its a complete new version but still sad that this issue was allready fixes in several versions before this update.
But Shit happens so i roll back till the new version is out ;-)

Haidiye00, 2025-03-27:
Hello, nothing unexpected, a new version will be released this week~Sorry

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00