bambulab/BambuStudio #6205: Bambu Studio V2 AMS Lite Slots cannot be selected

jomojewi, 2025-03-26:
### Bambu Studio Version

V2

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 12.7.6

### Printer

Bambu Lab A1 with AMS Lite

### How to reproduce

1. Slice the Object
2. GoTo Print
3. Unable to Select a Third Party Filament in any Slot. Only Bambu Filaments can be selected

### Actual results

unable to print

### Expected results

is this a bug ?

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ThisMustbeFabricating, 2025-03-26:
Also experiencing this issue with Bambu Studio 2.0 on my current MacOS installation. I have to force the filament in Prepare by placing each filament in the correct order during slicing, there is no ability to assign colors in the Print menu.

jamesarm97, 2025-03-27:
Same here, use to be able to override the filament selection during the print send dialog. It shows all the slots and the check next to the current but doesn't let you click and change it without reslicing.

JohnSmithCYM, 2025-03-27:
We have confirmed this bug and will fix it soon in next hotfix.

Haidiye00, 2025-03-28:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

If you continue to use a Mac laptop, you can avoid this problem by following these steps：
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00