bambulab/BambuStudio #6210: The program does not see AMS plastic during calibration

Postlast, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

Bambu Lab P1S combo

### How to reproduce

Calibration -> Flow rate or Flow dynamics -> manual calibration. Unable to select AMS plastic. Version 1.10.2.76 does not have this problem.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

No

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-03-27:
https://github.com/bambulab/BambuStudio/issues/6156 Same question and it confirmed.Could you try to let AMS feed first and try again.We have known it is bug and we will fix it soon.

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab