bambulab/BambuStudio #6211: Flow Dynamics and Rate filament dropdown selection is broken and just shows presets

jskkujawa, 2025-03-26:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Additional system information

[image]

### Printer

P1S

### How to reproduce

See image

### Actual results

can only select presets

### Expected results

can select filament

### Project file & Debug log uploads

not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jskkujawa, 2025-03-26:
Syncing device filament and rebooting bambu studio and PC did not fix.

All 4 ams slots are filled with filament

JohnSmithCYM, 2025-03-27:
https://github.com/bambulab/BambuStudio/issues/6156 Same question and it confirmed.Could you try to let AMS feed first and try again.We have known it is bug and we will fix it soon.

xsimtomx, 2025-03-27:
Can confirm that it _does_ work after first loading a spool into the AMS. I believe this wasn't the case pre-2.0.

zhimin-zeng-bambulab, 2025-03-31:
It has been fixed in [v02.00.01.50](https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50), please have a try.

closed 2025-03-31 by zhimin-zeng-bambulab