bambulab/BambuStudio #6212: Bambu Studio keep crashing on MacOS

Ibrahim-Albrahim, 2025-03-26:
### Bambu Studio Version

2.00.00.95

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 15.3.2

### Additional system information

it keeps crashing once I run the application

### Printer

All

### How to reproduce

every time I run the application

### Actual results

crashing

### Expected results

not cashing

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-27:
Hello, can you use the evevything software to search for Bambustudio and provide the log folder inside,thanks

[image]

GalenRhodes, 2025-03-27:
This is happening to me as well.  It seems to have something to do with it being Apple Silicon (M1 Pro Max) because it's crashing for the Mac version as well as the Windows version running on Windows 11 ARM under Parallels on my Mac.

[image]

GalenRhodes, 2025-03-27:
Here's a log from my computer if it helps.

[debug_Tue_Mar_25_15_20_08_2606.log.0.txt](https://github.com/user-attachments/files/19489052/debug_Tue_Mar_25_15_20_08_2606.log.0.txt)

[MacCrashDump.txt](https://github.com/user-attachments/files/19489057/MacCrashDump.txt)

GalenRhodes, 2025-03-27:
Actually, I take all of that back.  What I had to do was delete the config folder ~/Library/Application Support/BambuStudio and then it all worked!

Haidiye00, 2025-03-28:
[image]
Hello, try to find the configuration folder for BambuStudio on your Mac, delete it completely, and restart Bambustudio to try
find folder:BambuStudio

Ibrahim-Albrahim, 2025-03-28:
@Haidiye00 Worked fine now. thanks

closed 2025-03-28 by Ibrahim-Albrahim