bambulab/BambuStudio #6214: Studio 2.0 and H2D Filament Savings Not Calculated Correctly

MadMax389, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

10th gen Core i5, 32gb

### Printer

H2D

### How to reproduce

I've been getting used to the new features and noticed that the calculated filament saving using 2 heads and an AMS seem to be incorrect. The estimated savings does not match the actual saving when comparing slicing with 2 heads vs 1 head. 

To recreate:

1. Select H2D and add three cube primitives
2. Make each cube a different color and slice it with optimal settings
3. Note the total grams used and the estimated savings
4. Re-slice it with all three colors on one nozzle and compare

### Actual results

In my test case with 3 25 mm cubes, Studio estimated 120g total and estimated 118g savings over printing with one nozzle.

[image]

When sliced with one nozzle, the actual total is 191g, a savings of 71g, not 118g. Studio seems to overestimating the savings by 68%.

[image]

In another test Studio overestimated by 62%, 436g actual vs 707g estimated savings

[image]

[image]

### Expected results

Estimated savings vs actual savings should be within a few percent.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2025-03-27:
@MadMax389  Hello, please ensure that the flush values are the same when comparing. Could you please recheck the settings ? 

[image]

[image]

MadMax389, 2025-03-27:
My flush volume was set at .6. When I set it to 1.0, the savings estimate closely matches the actual savings. So I gather the estimated savings value does not account for the user's flush volume setting?

closed 2025-03-31 by XunZhangBambu