bambulab/BambuStudio #6215: No Preview on Bambustudio

Zer0Meow, 2025-03-27:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Built from source by yourself or others

### OS version

Gentoo Linux Kernel 6.12.16-gentoo-dist

### Additional system information

Cpu: Ryzen 9 5950x
Gpu: Rtx 3080

### Printer

a1 mini

### How to reproduce

1.Click on any model to print
2. look at preview

### Actual results

[image]

### Expected results

Showing the model in 3d

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19476993/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

guanyun-gudujian, 2025-03-30:
@Zer0Meow 
hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:

Image

if so, click the "Cube" item, then save the project, and send this project file to me, thanks.

Zer0Meow, 2025-04-08:
@guanyun-gudujian nothing pops up when doing a right click, ive tried several ways of installing bambu studio, i tried the flatpak and it just gives me a webkit-gtk error, i even tried forks like orcaslicer but i only get webkit-gtk errors from the binaries and i cant manage to compile it from an overlay. compiling bambu studio from an overlay is the only way to get it to launch but its useless since the preview doesnt do anything and it doesnt slice

Zer0Meow, 2025-04-13:
im sorry i didnt mean to close this issue it was an accident, either way what can i do to troubleshoot this? @guanyun-gudujian

g1bwt, 2025-04-14:
I am getting a similar problem with running the Flatpak version whereas the model will only display in Preview on the plate when using Orthogonal View rather than Perspective View (see attached screenshots please).[

[image]

[image]

](url)

First image with the spool model loaded but in Perspective View, the second image in Orthogonal View.

The BambuStudio Appimage version shows the model correctly.

BambuStudio Flatpak version 2.0.1.50

Operating System: Kubuntu 24.10
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.6.2
Kernel Version: 6.11.0-21-generic (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 3600 6-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: NV167
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: B450M H

jeremie-lesage, 2025-05-20:
Same issue with Bambu_Studio_linux_fedora-v02.00.03.54.AppImage

Operating System: Fedora 42
Windowing system: Wayland
CPU: i5-10300H
GPU: Nvidia RTX 2060

guanyun-gudujian, 2025-05-29:
@jeremie-lesage  @OswyChow   @g1bwt  @Zer0Meow 
seems a wayland only issue
i will try to fix it
pls see https://github.com/bambulab/BambuStudio/issues/6985

guanyun-gudujian, 2025-07-18:
@jeremie-lesage  @OswyChow  @g1bwt 

Hi, the offical package with the x11 mode workaround has been released

pls have a try

thanks

rember to remove all thoes workarounds manually setted before first

https://github.com/bambulab/BambuStudio/releases