bambulab/BambuStudio #6217: v2.0.0.95 Cannot select filament slot override on the print send screen

jamesarm97, 2025-03-27:
### Bambu Studio Version

v2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.3.1

### Printer

Bambu X1C

### How to reproduce

Prepare a model then click Print. The dialog box that shows before clicking the send button shows what filament will be used. The old version let you click the filament and override it on this screen to select a different AMS slot. This version shows all the slots and has the one currently checked but when I click on a different slot to override which filament I want to use on this screen, nothing happens. The old version would put a check mark next to the new slot and let you print using it.

### Actual results

When you click the filament showing on the print send screen, it drops down and shows all AMS slots and one with a check mark next to it. Clicking any other should move the check mark to it to use on this print but it does nothing.

### Expected results

Should be able to click on a filament and override it to a new AMS slot on the print send screen just like it use to work prior to v2

### Project file & Debug log uploads

N/A

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ThorsReign, 2025-03-27:
Same as https://github.com/bambulab/BambuStudio/issues/6202

bochinshen, 2025-03-27:
me too.
Bambu Studio Version

v2.0.0.95

OS version

MacOS 15.3.1

Printer

Bambu A1

Haidiye00, 2025-03-27:
Hello, if you have a Windows laptop, you can use the Windows version first to avoid this issue. We have fixed this issue and will release it as soon as possible. so sorry.

If you continue to use a Mac laptop, you can avoid this problem by following these steps：
Choose H2D preset -> Slice plate -> Print plate -> Choose another AMS slot( Just ignore the warning ) -> Go back to the Prepare page -> Switch to the preset that you want -> Slice again and print, then you can use AMS function normally.
If you reboot the software, you might need to do this again. Apologize for causing you inconvenience sincerely.

[image]

Haidiye00, 2025-03-28:
Hello, we have fixed this issue. Please download the latest version to use.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
We apologize for any inconvenience caused and will continue to improve the quality of our software.

closed 2025-03-28 by Haidiye00