bambulab/BambuStudio #6218: Flow Rate calibration not able to change filaments & AMS sync issue

Allen202012, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Go to the calibration tab of Bambu Studio, and go to the Flow rate calibration section.
2. Clicking manual calibration, the first issue would be present, which is unless there is a filament in the nozzle currently, AMS sync does not work whatsoever. The Filaments for calibration would be external filament, not AMS.
3. I have the plate setting to smooth PEI, not textured. Not sure if it affects it whatsoever though.
4. If you had loaded the filament previously before you chose it in the menu, the issue now occurs. The printer would try to load filament already in the nozzle, or in some cases load an entirely different color in another AMS slot.
5. The program would get stuck indefinitely in the "Heating nozzle" stage of filament loading, despite the nozzle being at the prerequisite temperature.
6. Then the only way to use the printer at all is to cancel the calibration as it's softlocked and unable to finish the calibration pieces. 

### Actual results

[image]

### Expected results

Well, it should be calibrating in the first place and not softlocked trying to do an action that is already the case.

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/19478392/New.Compressed.zipped.Folder.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-03-27:
https://github.com/bambulab/BambuStudio/issues/6156 Same quesiton, you can feed in the AMS first and do flow rate cali again.We will fix that in next version.

Allen202012, 2025-03-27:
[image]

[image]

[image]

Filament has already been feed into the AMS before starting cali, however in 6 mins it is still stuck in the nozzle heating stage and won't continue any longer.

JavaScriptJohn, 2025-03-27:
Same problem on a P1S. In the rare chance it does get passed the mapping table it gets stuck at preheating indefinitely.

zhimin-zeng-bambulab, 2025-03-27:
Thanks for reporting this issue, we will fix it as soon as possible.

josser, 2025-03-27:
I can confirm the same issue, and most terrifying thing that this brake Orca, and even downgrade to 1.x doesn't help

JavaScriptJohn, 2025-03-27:
> I can confirm the same issue, and most terrifying thing that this brake Orca, and even downgrade to 1.x doesn't help

Can confirm after downgrading calibration no longer works.

wwwabbit, 2025-03-27:
Confirming same issue with my P1S.  Tried last night on the old version, got the Failed to get AMS mapping table error.  Upgraded to the new v2 software and same thing.

cesaratb, 2025-03-27:
Confirmed. Same issue on my P1S.

wwwabbit, 2025-03-28:
Bambu must be doing something with this, tried it again this afternoon and now the only option I have is the External spool for calibration.

JavaScriptJohn, 2025-03-28:
> Bambu must be doing something with this, tried it again this afternoon and now the only option I have is the External spool for calibration.

Mine was doing this yesterday. It's very flaky sometimes it does it and sometimes it doesn't.

cesaratb, 2025-03-28:
> Bambu must be doing something with this, tried it again this afternoon and now the only option I have is the External spool for calibration.

Same here.

JavaScriptJohn, 2025-03-28:
Bambu listened! My calibration was completed without issue today using the latest version of Bambu Studio (2.0.1.50)

closed 2025-03-28 by Allen202012