bambulab/BambuStudio #6221: unnecessary supports  not supporting anything

graeioume, 2025-03-27:
### Bambu Studio Version

2.0.0

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. "Slice Plate" for this file with "enable supports" on, tree(auto)

### Actual results

Supports are being generated where the overhangs are removed by "remove small overhangs". Even when I tried to paint "No support" areas for those areas using the Sphere tool, it still would create supports for those areas.
[image]

### Expected results

The supports should not reach up to the head of the character.  The overhangs on the face should be taken care of by the "remove small overhangs" feature.

### Project file & Debug log uploads

[unrequested support generation.zip](https://github.com/user-attachments/files/19479011/unrequested.support.generation.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-03-27:
@graeioume Thanks for your report. Some part of the overhangs on the head is totally floating so it's set to the type SHARPTAIL. Then "remove small overhang" is not useful. 
And I tried paiting "No Support Area" and it did work, maybe you could paint larger area and try again or use the support blocker.
[image]
[image]

NothFie, 2025-03-27:
By the way, For such small-sized models, it is recommended to reduce “Branch distance” and "Branch diameter"(such as to {1,1})

graeioume, 2025-03-28:
These are very helpful pieces of advice, thank you!
I had tried painting the area with the "no supports" sphere brush, with "On overhangs only" checkbox selected, but it was not selecting those spots.
For that matter, twice now, the "On overhangs only" checkbox has failed to prevent painting outside the overhangs. If I get more consistentcy I'll make another bug report. :)

closed 2025-03-28 by graeioume