bambulab/BambuStudio #6223: after update to new version of Bambu, If I use import the 3mf file from Fusion 360. it gives me an error.

aj1004, 2025-03-27:
### Bambu Studio Version

Version 2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sequoia 15.3.2

### Additional system information

Chip: Apple M3 Max
Memory: 36 GB

[image]

### Printer

Bambu X1 Carbon

### How to reproduce

Anytime when I import 3mf file to Bambu Studio 2.0.0.95

### Actual results

Not reading 3mf file.

### Expected results

It is really anoing. 

### Project file & Debug log uploads

[3mf file error message.3mf.zip](https://github.com/user-attachments/files/19479185/3mf.file.error.message.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-03-27:
[image]
hello, i tested it is ok.
Can you record a video and upload it? Thank you

Haidiye00, 2025-03-31:
Hello, our internal version has been optimized. Thank you for your discovery

closed 2025-04-16 by Haidiye00