bambulab/BambuStudio #6225: Print job gets stuck when using the external spool on an AMS-equipped printer

luder888, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro Version	24H2

### Printer

P1S

### How to reproduce

1. On an AMS-equipped printer, pre-fed an external spool into the hotend.
2. In Bambu Studio, prepare any random print job
3. Click Print plate
4. In the "Send print job" dialog, select the printer that already had external filament loaded
5. In the "Filament" section, select the external spool
6. Do NOT expand the Advanced Options. Leave the "Use AMS" option to On
7. Click "Send" to send the job to the printer

### Actual results

The printer will warm up the bed and nozzle to the specified temperature, but then just sits there indefinitely.

### Expected results

The printer should proceed with the print, since the external spool was already selected, it should not be necessary to again expand the "Advanced Options" to turn off the AMS.

BTW, in the previous version of Bambu Studio the "Use AMS" checkbox was in plain sight. In 2.0, it was moved under the "Advanced Options", requiring an extra click.

I understand that if the "Use AMS" option is turned off, the print will succeed. However, given that the option is now hidden inside a manual, merely selecting an external spool should be sufficient to bypass the AMS.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bruceyv83, 2025-03-29:
I think I have something similar. My print gets stuck at the same point but only when printing an existing model from the SD card. If I freshly slice it works fine.
I noticed that when selecting a model from the SD card that there was an exclamation mark next to the filament type, it still allows me to select the correct filament but fails to prints.
I then noticed a tool tip (only appears on my laptop, not my desktop) with more details - see attached (ignore the printer is busy bit)
When it does eventually give up trying to print I get error code: 07ff00000020008

[image]

Krtzbrg, 2025-03-30:
Encounterin the same problem on 2.0.0.95 
Was using AMS, plugged in an external spool for the first time. 
Flushing filament worked but then even after restarting the Printer was stuck on preheating heatbed. 

Unplugging the AMS solved the problem for now. 
Try to reproduce it after my print is finished.

walterwongbbl, 2025-04-09:
Hi. @luder888 @bruceyv83 @Krtzbrg 
The use of the AMS option has caused a lot of confusion. We have received feedback and will remove this option, automatically deciding whether to use AMS.

luder888, 2025-04-09:
> Hi. [@luder888](https://github.com/luder888) [@bruceyv83](https://github.com/bruceyv83) [@Krtzbrg](https://github.com/Krtzbrg) The use of the AMS option has caused a lot of confusion. We have received feedback and will remove this option, automatically deciding whether to use AMS.

This particular bug I originally reported was resolved in the next version (2.0.1.50). This bug was duplicated by https://github.com/bambulab/BambuStudio/issues/6179 and https://github.com/bambulab/BambuStudio/issues/6203 and therefore can be marked as resolved.