bambulab/BambuStudio #6226: 输入坐标回车确认无效

SWeetT2, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2

### Printer

P1S

### How to reproduce

1.选择对象
2.点击工具栏“移动”
3.手动输入XYZ任意一个坐标
4.点击Enter

### Actual results

手动更改坐标后，单击Enter键并不会生效；实际需要我鼠标单击其他的输入框（例如我在X的坐标输入框输入100后，需要单击Y或者Z的输入框）或者是移动窗口的其他位置才能使坐标的更改生效。之后我发现单击打印板上的模型也能生效。

### Expected results

我希望在我手动输入坐标数据后，单击Enter键就能够使更改生效。我记得在之前的版本都是这样的。

### Project file & Debug log uploads

[logandfile.zip](https://github.com/user-attachments/files/19481355/logandfile.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

guanyun-gudujian, 2025-04-16:
@SWeetT2 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

SWeetT2, 2025-04-16:
Thank you, I installed the latest version, the problem has been fixed

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年4月16日 08:57 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***>,
***@***.***> |
| 主题 | Re: [bambulab/BambuStudio] 输入坐标回车确认无效 (Issue #6226) |

@SWeetT2
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

—

@SWeetT2
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

closed 2025-04-16 by SWeetT2