bambulab/BambuStudio #6227: 模型打印一半自己停了，两次了同一个模型

aichy126, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab github releases

### OS version

mac OS 15.3.2 

### Additional system information

[bug.3mf.zip](https://github.com/user-attachments/files/19481650/bug.3mf.zip)
2025-03-27 15:41:47.542382[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:48.542039[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:49.541858[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:50.541689[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:51.347356[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:41:51.347987[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:41:51.543244[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:52.541581[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:53.407182[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:41:53.407425[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:41:53.544643[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:54.545454[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:55.429888[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:41:55.430475[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:41:55.541053[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:56.543610[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:57.542911[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:58.542517[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:41:59.385462[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:41:59.385861[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:41:59.542906[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:00.541200[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:01.398987[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:01.399345[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:01.542013[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:02.542345[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:03.431474[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:03.431911[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:03.543197[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:04.542971[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:05.543444[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:06.491795[Thread 0x0000000206c84840]:keep alive = 300298, count = 5683
2025-03-27 15:42:06.491895[Thread 0x0000000206c84840]:publish_json: {"pushing":{"command":"pushall","push_target":1,"sequence_id":"20177","version":1}} code: 0
2025-03-27 15:42:06.539836[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:06.593315[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:06.593687[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:07.543282[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:08.543474[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:08.667221[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:08.667876[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:09.542301[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:10.542689[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:11.542811[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:12.542950[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:12.597397[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:12.597958[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:13.541933[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:14.541882[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:14.611534[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:14.611959[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:15.542776[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:16.542088[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
2025-03-27 15:42:16.626747[Thread 0x0000000206c84840]:update_printer_preset_name 6489start update preset_name
2025-03-27 15:42:16.627248[Thread 0x0000000206c84840]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-03-27 15:42:17.542146[Thread 0x0000000206c84840]:update_model_task times: 0 instance_id <= 0
[... 111 lines cut ...]

### Printer

P1s ams

### How to reproduce

模型已提供

### Actual results

同一个模型打印两次都是中间停了

### Expected results

同一个模型打印两次都是中间停了

### Project file & Debug log uploads

[debug_Thu_Mar_27_12_05_56_47548.log.0.zip](https://github.com/user-attachments/files/19481704/debug_Thu_Mar_27_12_05_56_47548.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-27:
Hello, I have used your model to print once here. There was no interruption during the process and it's almost finished now.
[image]

aichy126, 2025-03-27:
@BoBoYu-1  ok .I will wait until version 2.0 becomes more stable and try upgrading again. Thanks

BoBoYu-1, 2025-03-27:
@aichy126 Can this question be closed now?

aichy126, 2025-03-27:
@BoBoYu-1  yes can close it now.

closed 2025-03-28 by BoBoYu-1