bambulab/BambuStudio #6228: BambuStudio 2 ->  MakerLab -> Code tab - cmd + a doesn't work correctly

rmilkowski, 2025-03-27:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4

### Printer

X1C

### How to reproduce

Maker Lab -> Parametric Model Maker -> Code tab: I can edit the code or use mouse to select part of a code, but standard cmd (or ctrl) + a to select all text doesn't work - instead when I press cmd+a it ends up selecting all text in the most left tab (with user profile name and links to Home, Recently Opened, etc.

### Actual results

Wrong text from wrong tab is selected

### Expected results

cmd+a when the Code tab is activated should select all code

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rmilkowski, 2025-03-27:
Also, even if I use mouse to select some code if I then press cmd+c it doesn't copy it - I had to right click it and select copy from the menu.

realzorro, 2025-04-01:
Thank you very much for your valuable suggestion. It has been forwarded to the development team and optimization will be arranged in the near future. Please continue to follow and support Makerlab.